alexforencich/verilog-pcie #2: Intel PCIe IP model

bluecmd, 2019-10-16:
Hi,

Would you be open to add support for the Intel PCIe IP cores? If so, I might be able to contribute that.

Do you have any idea of how one might best integrate a new pinout and behaviour? While the PCIe BFM probably can remain 99% intact (I assume?), the interfaces from the core are quite different.

If you're curious, the interfaces are described [here])(https://www.intel.com/content/dam/www/programmable/us/en/pdfs/literature/ug/archives/ug-01097-1.3.pdf) under "Key Interfaces".

alexforencich, 2019-10-16:
Certainly, more device support is always better.  I have a couple of DE5-Net boards and a couple of Cyclone 10 GX dev boards.  I also did some messing around with the PCIe interface on the DE5-Net some time ago, so I'm somewhat familiar with the boar and the interface.  I plan on supporting those at some point, along with at least the Virtex 7.  

Currently, I divide the interfaces into two categories: streaming interfaces that deal in raw PCIe TLPs, and Xilinx Ultrascale that has their own header format (doesn't match the standard TLP format, and fixed header size instead of variable based on address size).  I have been focusing on the ultrascale parts because those are the parts I'm working with at the moment, but supporting a more generic TLP interface is definitely something that I want to do at some point.  For those parts, I think it makes sense to build the cores to support some standard streaming interface, such as AXI, and then make shims to convert whatever the specific FPGA provides.  

So, there are a few pieces that are needed.  The first thing to do is look at PCIe cores for as many parts as documentation is available for and try to distill out the main features.  How is the main TLP interface handled - pure streaming, or something weird, such as a segmented interface?  If there is a segmented interface, can it be placed in a start-in-lane-0-only mode?  What sideband signals are present?  Where can the local bus, device, and function numbers be read out from?  Max read request and payload size fields?  Extended tag enable bit?  BAR hit mask?  Then we'll need to write shims to translate to a common interface (i.e. convert avalon to AXI) and a mux to split incoming read and write requests from completions and split requests based on the BAR and/or function, and a demux to merge outgoing traffic.  Most of these pieces are already written for the Ultrascale interface, so it's mainly a matter of modifying existing code.  

A device model just needs to inherit from pcie.Device or pcie.Function.  Currently, the only device models are the Ultrascale model (UltrascalePCIe) in pcie_us.py and the Ultrascale Plus model (UltrascalePlusPCIe) in pcie_usp.py (which itself borrows a lot of stuff from pcie_us.py).  I have a much simpler model with an AXI interface that's not committed yet that would probably be a more reasonable starting point.

alexforencich, 2021-11-03:
The first Intel device model for Stratix 10 GX/SX/TX/MX H-Tile/L-Tile has been added in commit https://github.com/alexforencich/cocotbext-pcie/commit/9f94c4b52cf316e74948812b641ed109315a14bf, and the corresponding shim in commit https://github.com/alexforencich/verilog-pcie/commit/5c5876ff1d378cb0cbcafc3c4ac4606f214f9cd7.

alexforencich, 2022-07-13:
A model for the P-Tile has been added in commit https://github.com/alexforencich/cocotbext-pcie/commit/ff89e00596597bd7414c18e0f893294ca87b4bdd, with the corresponding shim in commit https://github.com/alexforencich/verilog-pcie/commit/2a727e04f7d30a3306967c75ff45ecd5f38c3437