alexforencich/verilog-pcie #3: pcie_disable_fatal_err.sh not working

yakirgb, 2019-11-07:
Hi,
Our servers crashing many times with:
```
[ 5420.891904] {1}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 1
[ 5420.891949] {1}[Hardware Error]: event severity: fatal
[ 5420.891967] {1}[Hardware Error]:  Error 0, type: fatal
[ 5420.891984] {1}[Hardware Error]:   section_type: PCIe error
[ 5420.892002] {1}[Hardware Error]:   port_type: 4, root port
[ 5420.892020] {1}[Hardware Error]:   version: 3.0
[ 5420.892035] {1}[Hardware Error]:   command: 0x0547, status: 0x4010
[ 5420.892053] {1}[Hardware Error]:   device_id: 0000:5d:00.0
[ 5420.892070] {1}[Hardware Error]:   slot: 0
[ 5420.892082] {1}[Hardware Error]:   secondary_bus: 0x5e
[ 5420.892098] {1}[Hardware Error]:   vendor_id: 0x8086, device_id: 0x2030
[ 5420.892117] {1}[Hardware Error]:   class_code: 000406
[ 5420.892132] {1}[Hardware Error]:   bridge: secondary_status: 0x2000, control: 0x0003
[ 5420.892156] Kernel panic - not syncing: Fatal hardware error!
```
model: PowerEdge FC640
NVMe: Intel
OS: CentOS Linux release 7.4.1708

i tried to use pcie_disable_fatal_err.sh but it's not working with errors of:
```
line 33: "0x0547" & ~0x0100: syntax error: operand expected (error token is ""0x0547" & ~0x0100")
line 40: "0x012e" & ~0x0004: syntax error: operand expected (error token is ""0x012e" & ~0x0004")
```

Full output:
```
[root@foo001 ~]# bash -x pcie_disable_fatal_err.sh 88:00.0
+ dev=88:00.0
+ '[' -z 88:00.0 ']'
+ '[' '!' -e /sys/bus/pci/devices/88:00.0 ']'
+ dev=0000:88:00.0
+ '[' '!' -e /sys/bus/pci/devices/0000:88:00.0 ']'
++++ readlink /sys/bus/pci/devices/0000:88:00.0
+++ dirname ../../../devices/pci0000:85/0000:85:02.0/0000:88:00.0
++ basename ../../../devices/pci0000:85/0000:85:02.0
+ port=0000:85:02.0
+ '[' '!' -e /sys/bus/pci/devices/0000:85:02.0 ']'
+ echo 'Disabling fatal error reporting on port 0000:85:02.0...'
Disabling fatal error reporting on port 0000:85:02.0...
++ setpci -s 0000:85:02.0 COMMAND
+ cmd=0547
+ echo Command: 0547
Command: 0547
pcie_disable_fatal_err.sh: line 33: "0x0547" & ~0x0100: syntax error: operand expected (error token is ""0x0547" & ~0x0100")
+ setpci -s 0000:85:02.0 COMMAND=
setpci: Missing value.
Try `setpci --help' for more information.
++ setpci -s 0000:85:02.0 CAP_EXP+8.w
+ ctrl=012e
+ echo 'Device control:' 012e
Device control: 012e
pcie_disable_fatal_err.sh: line 40: "0x012e" & ~0x0004: syntax error: operand expected (error token is ""0x012e" & ~0x0004")
+ setpci -s 0000:85:02.0 CAP_EXP+8.w=
setpci: Missing value.
Try `setpci --help' for more information.
```

alexforencich, 2019-11-07:
Are you experiencing this issue with a design using the verilog-pcie code, or some other device?

The problem with the script could be an issue with an oddball or old version of bash.  Try replacing all of the 0x with 16# and see if that helps.

alexforencich, 2019-11-07:
Here's the script updated with that change, let me know if it works on the system in question:

    #!/bin/bash

    dev=$1

    if [ -z "$dev" ]; then
        echo "Error: no device specified"
        exit 1
    fi

    if [ ! -e "/sys/bus/pci/devices/$dev" ]; then
        dev="0000:$dev"
    fi

    if [ ! -e "/sys/bus/pci/devices/$dev" ]; then
        echo "Error: device $dev not found"
        exit 1
    fi

    port=$(basename $(dirname $(readlink "/sys/bus/pci/devices/$dev")))

    if [ ! -e "/sys/bus/pci/devices/$port" ]; then
        echo "Error: device $port not found"
        exit 1
    fi

    echo "Disabling fatal error reporting on port $port..."

    cmd=$(setpci -s $port COMMAND)

    echo "Command:" $cmd

    # clear SERR bit in command register
    setpci -s $port COMMAND=$(printf "%04x" $(("16#$cmd" & ~16#0100)))

    ctrl=$(setpci -s $port CAP_EXP+8.w)
     
    echo "Device control:" $ctrl

    # clear fatal error reporting enable bit in device control register
    setpci -s $port CAP_EXP+8.w=$(printf "%04x" $(("16#$ctrl" & ~16#0004)))

yakirgb, 2019-11-14:
Hi, still same error:
```
[root@foo001 ~]# bash -x  pcie_disable_fatal_err.sh 88:00.0
+ dev=88:00.0
+ '[' -z 88:00.0 ']'
+ '[' '!' -e /sys/bus/pci/devices/88:00.0 ']'
+ dev=0000:88:00.0
+ '[' '!' -e /sys/bus/pci/devices/0000:88:00.0 ']'
++++ readlink /sys/bus/pci/devices/0000:88:00.0
+++ dirname ../../../devices/pci0000:85/0000:85:02.0/0000:88:00.0
++ basename ../../../devices/pci0000:85/0000:85:02.0
+ port=0000:85:02.0
+ '[' '!' -e /sys/bus/pci/devices/0000:85:02.0 ']'
+ echo 'Disabling fatal error reporting on port 0000:85:02.0...'
Disabling fatal error reporting on port 0000:85:02.0...
++ setpci -s 0000:85:02.0 COMMAND
+ cmd=0547
+ echo Command: 0547
Command: 0547
pcie_disable_fatal_err.sh 33: "16#0547" & ~16#0100: syntax error: operand expected (error token is ""16#0547" & ~16#0100")
+ setpci -s 0000:85:02.0 COMMAND=
setpci: Missing value.
Try `setpci --help' for more information.
++ setpci -s 0000:85:02.0 CAP_EXP+8.w
+ ctrl=0124
+ echo 'Device control:' 0124
Device control: 0124
pcie_disable_fatal_err.sh: line 40: "16#0124" & ~16#0004: syntax error: operand expected (error token is ""16#0124" & ~16#0004")
+ setpci -s 0000:85:02.0 CAP_EXP+8.w=
setpci: Missing value.
```

Something wrong with 16# :
```[root@foo001 ~]# printf $(("16#"))
-bash: "16#": syntax error: operand expected (error token is ""16#"")

[root@foo001 ~]# /bin/printf --version
printf (GNU coreutils) 8.22
```
Thanks, Yakir.

alexforencich, 2019-11-14:
What version of bash are you running?

yakirgb, 2019-11-17:
[root@foo001 ~]# bash --version
GNU bash, version 4.2.46(2)-release (x86_64-redhat-linux-gnu)

alexforencich, 2019-11-17:
Alright, I downloaded a live ISO and did some poking around; it seems to be the quotes that are tripping it up.  Try this:

    #!/bin/bash

    dev=$1

    if [ -z "$dev" ]; then
        echo "Error: no device specified"
        exit 1
    fi

    if [ ! -e "/sys/bus/pci/devices/$dev" ]; then
        dev="0000:$dev"
    fi

    if [ ! -e "/sys/bus/pci/devices/$dev" ]; then
        echo "Error: device $dev not found"
        exit 1
    fi

    port=$(basename $(dirname $(readlink "/sys/bus/pci/devices/$dev")))

    if [ ! -e "/sys/bus/pci/devices/$port" ]; then
        echo "Error: device $port not found"
        exit 1
    fi

    echo "Disabling fatal error reporting on port $port..."

    cmd=$(setpci -s $port COMMAND)

    echo "Command:" $cmd

    # clear SERR bit in command register
    setpci -s $port COMMAND=$(printf "%04x" $((0x$cmd & ~0x0100)))

    ctrl=$(setpci -s $port CAP_EXP+8.w)
     
    echo "Device control:" $ctrl

    # clear fatal error reporting enable bit in device control register
    setpci -s $port CAP_EXP+8.w=$(printf "%04x" $((0x$ctrl & ~0x0004)))

yakirgb, 2019-11-17:
Great, it's working, going to test it on few servers.

alexforencich, 2019-11-17:
Excellent; I will update all the scripts to remove those quotes

yakirgb, 2019-11-18:
Servers crashing also after the fix, maybe it's other issue.
```
[ 1831.545200] {1}[Hardware Error]: Hardware error from APEI Generic Hardware Error Source: 5
[ 1831.545343] {1}[Hardware Error]: event severity: fatal
[ 1831.545421] {1}[Hardware Error]:  Error 0, type: fatal
[ 1831.545497] {1}[Hardware Error]:   section_type: PCIe error
[ 1831.545582] {1}[Hardware Error]:   port_type: 4, root port
[ 1831.545663] {1}[Hardware Error]:   version: 3.0
[ 1831.545729] {1}[Hardware Error]:   command: 0x0547, status: 0x4010
[ 1831.545831] {1}[Hardware Error]:   device_id: 0000:5d:00.0
[ 1831.545912] {1}[Hardware Error]:   slot: 0
[ 1831.545973] {1}[Hardware Error]:   secondary_bus: 0x5e
[ 1831.546046] {1}[Hardware Error]:   vendor_id: 0x8086, device_id: 0x2030
[ 1831.546143] {1}[Hardware Error]:   class_code: 000406
[ 1831.546221] {1}[Hardware Error]:   bridge: secondary_status: 0x2000, control: 0x0003
[ 1831.546331] Kernel panic - not syncing: Fatal hardware error!
```

alexforencich, 2019-11-24:
That's interesting.  I wonder if something (i.e. iDRAC) is setting those bits again.  I just ran the exact same script on a Dell R540 server to prevent the server from rebooting when reloading an FPGA.  What is triggering this error on your machines?  And what does device 0000:5d:00.0 correspond to?  Apparently, at least according to that error message, that device has the SERR bit set; maybe the error reporting bits on that device need to be cleared.

yakirgb, 2019-11-24:
Hi, @alexforencich:
54.00.0 device:
```
5d:00.0 PCI bridge: Intel Corporation Sky Lake-E PCI Express Root Port A (rev 04) (prog-if 00 [Normal decode])
	Physical Slot: 2
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr+ Stepping- SERR+ FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 29
	NUMA node: 0
	Bus: primary=5d, secondary=5e, subordinate=5f, sec-latency=0
	I/O behind bridge: 0000f000-00000fff
	Memory behind bridge: b8800000-b88fffff
	Prefetchable memory behind bridge: 00000000fff00000-00000000000fffff
	Secondary status: 66MHz- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort+ <SERR- <PERR-
	BridgeCtl: Parity+ SERR+ NoISA- VGA- MAbort- >Reset- FastB2B-
		PriDiscTmr- SecDiscTmr- DiscTmrStat- DiscTmrSERREn-
	Capabilities: [40] Subsystem: Intel Corporation Device 0000
	Capabilities: [60] MSI: Enable+ Count=1/2 Maskable+ 64bit-
		Address: fee00038  Data: 0000
		Masking: 00000002  Pending: 00000000
	Capabilities: [90] Express (v2) Root Port (Slot-), MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0
			ExtTag+ RBE+
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal+ Unsupported-
			RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
			MaxPayload 128 bytes, MaxReadReq 128 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #9, Speed 8GT/s, Width x16, ASPM L1, Exit Latency L0s <512ns, L1 <16us
			ClockPM- Surprise+ LLActRep+ BwNot+ ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s, Width x16, TrErr- Train- SlotClk+ DLActive+ BWMgmt+ ABWMgmt-
		RootCtl: ErrCorrectable- ErrNon-Fatal+ ErrFatal+ PMEIntEna+ CRSVisible+
		RootCap: CRSVisible+
		RootSta: PME ReqID 0000, PMEStatus- PMEPending-
		DevCap2: Completion Timeout: Range BCD, TimeoutDis+, LTR-, OBFF Not Supported ARIFwd+
		DevCtl2: Completion Timeout: 65ms to 210ms, TimeoutDis-, LTR-, OBFF Disabled ARIFwd-
		LnkCtl2: Target Link Speed: 8GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete+, EqualizationPhase1+
			 EqualizationPhase2+, EqualizationPhase3+, LinkEqualizationRequest-
	Capabilities: [e0] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D3 NoSoftRst+ PME-Enable+ DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Vendor Specific Information: ID=0002 Rev=0 Len=00c <?>
	Capabilities: [110 v1] Access Control Services
		ACSCap:	SrcValid+ TransBlk+ ReqRedir+ CmpltRedir+ UpstreamFwd+ EgressCtrl- DirectTrans-
		ACSCtl:	SrcValid- TransBlk- ReqRedir- CmpltRedir- UpstreamFwd- EgressCtrl- DirectTrans-
	Capabilities: [148 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt+ RxOF- MalfTLP- ECRC- UnsupReq+ ACSViol+
		UESvrt:	DLP+ SDES+ TLP+ FCP+ CmpltTO+ CmpltAbrt+ UnxCmplt- RxOF+ MalfTLP+ ECRC+ UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr+ BadTLP+ BadDLLP+ Rollover+ Timeout+ NonFatalErr+
		AERCap:	First Error Pointer: 05, GenCap+ CGenEn+ ChkCap+ ChkEn+
	Capabilities: [1d0 v1] Vendor Specific Information: ID=0003 Rev=1 Len=00a <?>
	Capabilities: [250 v1] #19
	Capabilities: [280 v1] Vendor Specific Information: ID=0005 Rev=3 Len=018 <?>
	Capabilities: [300 v1] Vendor Specific Information: ID=0008 Rev=0 Len=038 <?>
	Kernel driver in use: pcieport
```

alexforencich, 2019-11-24:
so 5d:00.0 is a root port; what devices are under that port?