alexforencich/verilog-pcie #4: Package tb/pcie.py as a pip package

bluecmd, 2019-11-12:
Hi,

I'm starting to figure out how I want to do my testbenches for my project - and your pcie.py is really impressive and useful. It seems to be standalone to the point where that's the only thing one would need to simulate PCIe transactions, and in my case the only thing I would have to use I think.

That made me think: Would you be open for breaking that file (and potentially the AXI and US stuff as well, although I don't need that myself right now) out to its own library and make it a pip3 module? That way we can both use it and patches would go to a singular place.

alexforencich, 2019-11-13:
Probably not going to do that.  Occasionally I make extensive, breaking changes to these files, and keeping everything in the same repo means there is no chance for any versioning issues with external components.  Especially for pcie.py - I intend on extending that down the stack, hopefully all the way to the PIPE interface at some point, and that could possibly involve significant changes to the interface.  As it stands right now, I can make breaking changes and deal with fixing things later when I update the PCIe library in repos that use it.  This is not so much about this repo, but projects that include this code as a subtree or submodule.

bluecmd, 2019-11-13:
That's ok, thanks for the consideration though.

Would you mind it if I forked it and packaged it?

alexforencich, 2019-11-13:
It's MIT licensed, so you can do whatever you want with it.  However, this repository will be authoritative.

bluecmd, 2019-11-14:
SGTM, thanks!

closed 2019-11-14 by bluecmd