alexforencich/verilog-pcie #5: Make failure for example VU1525/fpga_axi/driver

FPGA-Bot-Yang, 2020-02-12:
Hi Alex,

First thanks for provide this source code!

I'm looking for a linux host driver for FPGA PCIe. I was trying to use your driver under example/VU1525/fpga_axi/driver. When I make under that path, I encounter the error stating: **implicit declaration of function ‘pci_alloc_irq_vectors'**. I'm wondering if this can be solved by adding extra library in the include section. The full error message is attached below:

$ make
make: Warning: File 'Makefile' has modification time 593811147 s in the future
make -C /lib/modules/3.10.0-693.2.2.el7.x86_64/build M=/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver modules
make[1]: Entering directory '/usr/src/kernels/3.10.0-693.2.2.el7.x86_64'
make[1]: Warning: File 'arch/x86/Makefile' has modification time 518859695 s in the future
make[2]: Warning: File 'scripts/Makefile.lib' has modification time 518856459 s in the future
  CC [M]  /AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.o
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c: In function ‘edev_probe’:
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:134:11: error: implicit declaration of function ‘pci_alloc_irq_vectors’ [-Werror=implicit-function-declaration]
     ret = pci_alloc_irq_vectors(pdev, 1, 32, PCI_IRQ_MSI);
           ^~~~~~~~~~~~~~~~~~~~~
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:134:46: error: ‘PCI_IRQ_MSI’ undeclared (first use in this function)
     ret = pci_alloc_irq_vectors(pdev, 1, 32, PCI_IRQ_MSI);
                                              ^~~~~~~~~~~
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:134:46: note: each undeclared identifier is reported only once for each function it appears in
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:142:11: error: implicit declaration of function ‘pci_request_ir ’ [-Werror=implicit-function-declaration]
     ret = pci_request_irq(pdev, 0, edev_intr, 0, edev, "edev");
           ^~~~~~~~~~~~~~~
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:230:5: error: implicit declaration of function ‘pci_free_irq_vectors’ [-Werror=implicit-function-declaration]
     pci_free_irq_vectors(pdev);
     ^~~~~~~~~~~~~~~~~~~~
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c: In function ‘edev_remove’:
/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.c:254:5: error: implicit declaration of function ‘pci_free_irq’ [-Werror=implicit-function-declaration]
     pci_free_irq(pdev, 0, edev);
     ^~~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:341: recipe for target '/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.o' failed
make[2]: *** [/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver/example_driver.o] Error 1
Makefile:1305: recipe for target '_module_/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver' failed
make[1]: *** [_module_/AlexForencich/verilog-pcie/example/VCU1525/fpga_axi/driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/3.10.0-693.2.2.el7.x86_64'
Makefile:7: recipe for target 'all' failed
make: *** [all] Error 2

FPGA-Bot-Yang, 2020-02-12:
FYI, my current included libraries in **exmaple_driver.c** are:

#include "example_driver.h"
#include <linux/module.h>
#include <linux/pci.h>
#include <linux/pci-aspm.h>
#include <linux/delay.h>
#include <linux/irq.h>

alexforencich, 2020-02-12:
Sounds like you're running a rather old version of Linux. The API is a bit of a moving target. What OS are you running, and what's the exact kernel version?

FPGA-Bot-Yang, 2020-02-12:
Hi Alex,

My operation system information is below:

NAME="Red Hat Enterprise Linux Server"
VERSION="7.4 (Maipo)"
ID="rhel"
ID_LIKE="fedora"
VARIANT="Server"
VARIANT_ID="server"
VERSION_ID="7.4"
PRETTY_NAME="Red Hat Enterprise Linux Server 7.4 (Maipo)"
ANSI_COLOR="0;31"
CPE_NAME="cpe:/o:redhat:enterprise_linux:7.4:GA:server"
HOME_URL="https://www.redhat.com/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"

REDHAT_BUGZILLA_PRODUCT="Red Hat Enterprise Linux 7"
REDHAT_BUGZILLA_PRODUCT_VERSION=7.4
REDHAT_SUPPORT_PRODUCT="Red Hat Enterprise Linux"
REDHAT_SUPPORT_PRODUCT_VERSION="7.4"

Kernel version is: 
3.10.0-693.2.2.el7.x86_64

alexforencich, 2020-02-12:
3.10? That is absolutely ancient. A number of kernel APIs have changed since then, including apparently a bunch of the MSI related ones. Those will have to be reimplemented in whatever the equivalent was in 3.10, and selected based on kernel version with preprocessor directives.

FPGA-Bot-Yang, 2020-02-12:
I see. Which version do you suggest that should be used in order to run this design seamlessly?

alexforencich, 2020-02-12:
Well, there's basically two options: adapt the driver so that it can use the 3.10 kernel API under RHEL, or switch the OS to something more recent like Ubuntu 16.04 or 18.04 LTS.

FPGA-Bot-Yang, 2020-02-12:
Thanks for the information, Alex! Much appreciated!

closed 2020-02-12 by FPGA-Bot-Yang