alexforencich/verilog-pcie #7: vivado version

shahbaazlokh, 2020-06-18:
i tried to compile vcu1525 example design using makefile using vivado 2019.2, but it fails to generate bitstream, which vivado version should i use to make project?

thanks

alexforencich, 2020-06-18:
I have been using 2019.1 up until now.  However, I just got set up with a current license, so I will most likely be migrating to 2020.1.  Odd that you get that particular error; I'll try it myself in 2019.2 and see what happens.

shahbaazlokh, 2020-06-18:
ultrascale+ pcie is is configured as gen3 x2, i think if i just comment out other pcie_tx_* and pcie_rx_* pins from xdc files this should solve this issue, will try to comment out and generate bitstream again, also need to make changes in top level wrapper in fpga.v file to 
    input  wire [1:0] pcie_rx_p,
    input  wire [1:0] pcie_rx_n,
    output wire [1:0] pcie_tx_p,
    output wire [1:0] pcie_tx_n,

alexforencich, 2020-06-18:
Ah yeah, if you don't use all the lanes, then you need to comment out or remove the unused ports in fpga.v and possibly fpga.xdc.  If you do that, then I think it should synthesize just fine.

shahbaazlokh, 2020-06-18:
i was able to make bitstream, but it did not enumerated on my host, doing lspci gives no xilinx devices,
also tried 
sudo lspci -vvv -d 10ee:

alexforencich, 2020-06-18:
Did you configure the PCIe hard IP core with VID 10ee?

shahbaazlokh, 2020-06-18:
yes

alexforencich, 2020-06-18:
OK, so this is a problem with the hard IP core.  The hard IP core is responsible for enumeration, the user logic sitting behind it is not used until the hard IP core is enumerated by the OS and a driver is attached.  I know there are two ref clk sites for PCIe, maybe the other one is required for use with less than 16 lanes.  Try changing the ref clk site and see if that helps.  Although presumably you would get an angry error message if Vivado didn't like the ref clk assignment.  And did you do a full reboot of the host machine after configuring the FPGA?  Generally you'll need to do that when the PCIe configuration changes.

shahbaazlokh, 2020-06-18:
yes, i did full reboot, i also made mcs file and proframmed flash just to make sure current bitstream is loaded by doing a cold shut-down and also waited for capacitors to discharge out for few minutes.

though i will try a different pcie clock pins, thank you for your quick response.

SeDi343, 2020-10-19:
Sorry for asking. But what makefile are you talking about? I can't find any in this project :S

alexforencich, 2020-10-19:
https://github.com/alexforencich/verilog-pcie/blob/master/example/VCU1525/fpga_axi/fpga/Makefile