alexforencich/verilog-pcie #8: Request for U200 Support

CGavinMullis, 2020-07-17:
I noticed that you just recently added U280 support and U50 support to your project. I wanted to know if you were in the process of adding U200 support or what we could do to migrate either your U50 or U280 example code to our U200 board. I see that your pcie4 tcl script for the U280 example has some parameters that are too high bit-wise for the U200 pcie4. Is it just as simple as modifying this tcl script and keeping everything else the same?

Any help is appreciated, thanks!

alexforencich, 2020-07-17:
I don't have access to a U200.  When I get one, I will add support for it, but there is no timeline for that at the moment.  If you want to accelerate that by donating hardware, then that can be arranged.  

You may be able to directly port the VCU1525 design to the U200 by changing the device and the pin assignments.  I think the VCU1525/AU200/AU250 boards have a very similar design.

CGavinMullis, 2020-07-17:
I appreciate the quick response. We will look into porting over the VCU1525 design and update you if we have any luck.

hl2007335, 2020-07-17:
sorry，i just add it to  my project，and don't do anything to it。 I'm just a green hander for pcie 4

hl2007335, 2020-07-17:
sorry，i just add it to  my project，and don't do anything to it。 I'm just a green hander for pcie

alexforencich, 2020-09-18:
Your wish is my command: https://github.com/alexforencich/verilog-pcie/tree/master/example/AU200/fpga_axi

Also, I added support for the U250: https://github.com/alexforencich/verilog-pcie/tree/master/example/AU250/fpga_axi

(although it helps that Xilinx donated a U200 and a U250)

Let me know how it works for you.

closed 2020-10-06 by alexforencich