alexforencich/verilog-pcie #9: AU50 example fails to generate bitstream

TaekyungHeo, 2020-07-24:
Hello, Alex.

I have tried to build [the AU50 example](https://github.com/alexforencich/verilog-pcie/tree/master/example/AU50/fpga_axi), but it fails to generate bitstream with the following messages.

DRC complains about "Unspecified I/O Standard" and "Unconstrained Logical Port".

```
$ make
...
source generate_bit.tcl
# open_project fpga.xpr
Scanning sources...
Finished scanning sources
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/tools/Xilinx/Vivado/2020.1/data/ip'.
open_project: Time (s): cpu = 00:00:04 ; elapsed = 00:00:05 . Memory (MB): peak = 2158.355 ; gain = 2.016 ; free physical = 247677 ; free virtual = 255592
# open_run impl_1
INFO: [Device 21-403] Loading part xcu50-fsvh2104-2-e
Netlist sorting complete. Time (s): cpu = 00:00:00.57 ; elapsed = 00:00:00.58 . Memory (MB): peak = 2906.109 ; gain = 0.000 ; free physical = 246452 ; free virtual = 254366
INFO: [Netlist 29-17] Analyzing 670 Unisim elements for replacement
INFO: [Netlist 29-28] Unisim Transformation completed in 0 CPU seconds
INFO: [Project 1-479] Netlist was created with Vivado 2020.1
INFO: [Project 1-570] Preparing netlist for logic optimization
INFO: [Timing 38-478] Restoring timing data from binary archive.
INFO: [Timing 38-479] Binary timing data restore complete.
INFO: [Project 1-856] Restoring constraints from binary archive.
INFO: [Project 1-853] Binary constraint restore complete.
Reading XDEF placement.
Reading placer database...
Reading XDEF routing.
Read XDEF File: Time (s): cpu = 00:00:04 ; elapsed = 00:00:05 . Memory (MB): peak = 3627.258 ; gain = 102.719 ; free physical = 245854 ; free virtual = 253769
Restored from archive | CPU: 4.690000 secs | Memory: 72.418594 MB |
Finished XDEF File Restore: Time (s): cpu = 00:00:05 ; elapsed = 00:00:05 . Memory (MB): peak = 3627.258 ; gain = 102.719 ; free physical = 245854 ; free virtual = 253769
Netlist sorting complete. Time (s): cpu = 00:00:00 ; elapsed = 00:00:00 . Memory (MB): peak = 3627.258 ; gain = 0.000 ; free physical = 245861 ; free virtual = 253776
INFO: [Project 1-111] Unisim Transformation Summary:
  A total of 152 instances were transformed.
  IBUF => IBUF (IBUFCTRL, INBUF): 1 instance
  RAM256X1D => RAM256X1D (MUXF7(x4), MUXF8(x2), RAMD64E(x8)): 54 instances
  RAM256X1S => RAM256X1S (MUXF7(x2), MUXF8, RAMS64E(x4)): 2 instances
  RAM32M16 => RAM32M16 (RAMD32(x14), RAMS32(x2)): 10 instances
  RAM32X1D => RAM32X1D (RAMD32(x2)): 1 instance
  RAM64M8 => RAM64M8 (RAMD64E(x8)): 72 instances
  RAM64X1D => RAM64X1D (RAMD64E(x2)): 12 instances

open_run: Time (s): cpu = 00:00:37 ; elapsed = 00:00:52 . Memory (MB): peak = 3627.258 ; gain = 1468.902 ; free physical = 245861 ; free virtual = 253776
# write_bitstream -force fpga.bit
Command: write_bitstream -force fpga.bit
Attempting to get a license for feature 'Implementation' and/or device 'xcu50'
INFO: [Common 17-349] Got license for feature 'Implementation' and/or device 'xcu50'
Running DRC as a precondition to command write_bitstream
INFO: [IP_Flow 19-1839] IP Catalog is up to date.
INFO: [DRC 23-27] Running DRC with 8 threads
ERROR: [DRC NSTD-1] Unspecified I/O Standard: 3 out of 71 logical ports use I/O standard (IOSTANDARD) value 'DEFAULT', instead of a user assigned specific value. This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all I/O standards. This design will fail to generate a bitstream unless all logical ports have a user specified I/O standard value defined. To allow bitstream creation with unspecified I/O standard values (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks NSTD-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run. Problem ports: qsfp_led_stat_y, qsfp_led_act, and qsfp_led_stat_g.
ERROR: [DRC UCIO-1] Unconstrained Logical Port: 3 out of 71 logical ports have no user assigned specific location constraint (LOC). This may cause I/O contention or incompatibility with the board power or connectivity affecting performance, signal integrity or in extreme cases cause damage to the device or the components to which it is connected. To correct this violation, specify all pin locations. This design will fail to generate a bitstream unless all logical ports have a user specified site LOC constraint defined.  To allow bitstream creation with unspecified pin locations (not recommended), use this command: set_property SEVERITY {Warning} [get_drc_checks UCIO-1].  NOTE: When using the Vivado Runs infrastructure (e.g. launch_runs Tcl command), add this command to a .tcl file and add that file as a pre-hook for write_bitstream step for the implementation run.  Problem ports: qsfp_led_stat_y, qsfp_led_act, and qsfp_led_stat_g.
INFO: [DRC REQP-1858] RAMB36E2_writefirst_collision_advisory: Synchronous clocking is detected for BRAM (pcie4c_uscale_plus_inst/inst/pcie_4_0_pipe_inst/pcie_4_0_bram_inst/RAM32K.bram_comp_inst/bram_16k_0_int/ECC_RAM.RAMB36E2[0].ramb36e2_inst) in SDP mode with WRITE_FIRST write-mode. It is strongly suggested to change this mode to NO_CHANGE for best power characteristics. However, both WRITE_FIRST and NO_CHANGE may exhibit address collisions if the same address appears on both read and write ports resulting in unknown or corrupted read data. It is suggested to confirm via simulation that an address collision never occurs and if so it is suggested to try and avoid this situation. If address collisions cannot be avoided, the write-mode may be set to READ_FIRST which guarantees that the read data is the prior contents of the memory at the cost of additional power in the design. See the FPGA Memory Resources User Guide for additional information.
INFO: [DRC REQP-1858] RAMB36E2_writefirst_collision_advisory: Synchronous clocking is detected for BRAM (pcie4c_uscale_plus_inst/inst/pcie_4_0_pipe_inst/pcie_4_0_bram_inst/RAM32K.bram_comp_inst/bram_16k_0_int/ECC_RAM.RAMB36E2[1].ramb36e2_inst) in SDP mode with WRITE_FIRST write-mode. It is strongly suggested to change this mode to NO_CHANGE for best power characteristics. However, both WRITE_FIRST and NO_CHANGE may exhibit address collisions if the same address appears on both read and write ports resulting in unknown or corrupted read data. It is suggested to confirm via simulation that an address collision never occurs and if so it is suggested to try and avoid this situation. If address collisions cannot be avoided, the write-mode may be set to READ_FIRST which guarantees that the read data is the prior contents of the memory at the cost of additional power in the design. See the FPGA Memory Resources User Guide for additional information.
INFO: [DRC REQP-1858] RAMB36E2_writefirst_collision_advisory: Synchronous clocking is detected for BRAM (pcie4c_uscale_plus_inst/inst/pcie_4_0_pipe_inst/pcie_4_0_bram_inst/RAM32K.bram_comp_inst/bram_16k_0_int/ECC_RAM.RAMB36E2[2].ramb36e2_inst) in SDP mode with WRITE_FIRST write-mode. It is strongly suggested to change this mode to NO_CHANGE for best power characteristics. However, both WRITE_FIRST and NO_CHANGE may exhibit address collisions if the same address appears on both read and write ports resulting in unknown or corrupted read data. It is suggested to confirm via simulation that an address collision never occurs and if so it is suggested to try and avoid this situation. If address collisions cannot be avoided, the write-mode may be set to READ_FIRST which guarantees that the read data is the prior contents of the memory at the cost of additional power in the design. See the FPGA Memory Resources User Guide for additional information.
INFO: [DRC REQP-1858] RAMB36E2_writefirst_collision_advisory: Synchronous clocking is detected for BRAM (pcie4c_uscale_plus_inst/inst/pcie_4_0_pipe_inst/pcie_4_0_bram_inst/RAM32K.bram_comp_inst/bram_16k_0_int/ECC_RAM.RAMB36E2[3].ramb36e2_inst) in SDP mode with WRITE_FIRST write-mode. It is strongly suggested to change this mode to NO_CHANGE for best power characteristics. However, both WRITE_FIRST and NO_CHANGE may exhibit address collisions if the same address appears on both read and write ports resulting in unkno [... cut ...]
write_bitstream: Time (s): cpu = 00:00:24 ; elapsed = 00:00:11 . Memory (MB): peak = 4017.777 ; gain = 390.520 ; free physical = 245743 ; free virtual = 253658
ERROR: [Common 17-39] 'write_bitstream' failed due to earlier errors.

    while executing
"write_bitstream -force fpga.bit"
    (file "generate_bit.tcl" line 3)
INFO: [Common 17-206] Exiting Vivado at Fri Jul 24 14:14:00 2020...
../common/vivado.mk:110: recipe for target 'fpga.bit' failed
make[1]: *** [fpga.bit] Error 1
make[1]: Leaving directory '/home/tkheo/verilog-pcie/example/AU50/fpga_axi/fpga'
Makefile:14: recipe for target 'fpga' failed
make: *** [fpga] Error 2
```

alexforencich, 2020-07-24:
Looks like a couple of lines were commented out in the XDC file.  Try it again and let me know if you run in to any other issues.

TaekyungHeo, 2020-07-24:
Thanks! It works.

closed 2020-07-24 by TaekyungHeo