alexforencich/verilog-pcie #14: Module not found issue comming

joychauhan-sudo, 2021-03-16:
Hello team, 

Tried to run test after making vivado project. while running make in tb/fpga_core folder below error is coming. 

make results.xml
make[1]: Entering directory '/media/joy/D/cocotb/cocotb_work/cocotbext-pcie/verilog-pcie/example/ZCU106/fpga_axi/tb/fpga_core'
mkdir -p sim_build
/usr/bin/iverilog -o sim_build/sim.vvp -D COCOTB_SIM=1 -s fpga_core -P fpga_core.AXIS_PCIE_DATA_WIDTH=128 -P fpga_core.AXIS_PCIE_KEEP_WIDTH=4 -P fpga_core.AXIS_PCIE_RQ_USER_WIDTH=62 -P fpga_core.AXIS_PCIE_RC_USER_WIDTH=75 -P fpga_core.AXIS_PCIE_CQ_USER_WIDTH=88 -P fpga_core.AXIS_PCIE_CC_USER_WIDTH=33 -P fpga_core.RQ_SEQ_NUM_WIDTH=6 -f sim_build/cmds.f -g2012   ../../rtl/fpga_core.v ../../rtl/axi_ram.v ../../rtl/axis_register.v ../../lib/pcie/rtl/axis_arb_mux.v ../../lib/pcie/rtl/pcie_us_axil_master.v ../../lib/pcie/rtl/pcie_us_axi_dma.v ../../lib/pcie/rtl/pcie_us_axi_dma_rd.v ../../lib/pcie/rtl/pcie_us_axi_dma_wr.v ../../lib/pcie/rtl/pcie_us_axi_master.v ../../lib/pcie/rtl/pcie_us_axi_master_rd.v ../../lib/pcie/rtl/pcie_us_axi_master_wr.v ../../lib/pcie/rtl/pcie_us_axis_cq_demux.v ../../lib/pcie/rtl/pcie_us_cfg.v ../../lib/pcie/rtl/pcie_us_msi.v ../../lib/pcie/rtl/arbiter.v ../../lib/pcie/rtl/priority_encoder.v ../../lib/pcie/rtl/pulse_merge.v
:0: warning: parameter RQ_SEQ_NUM_WIDTH not found in fpga_core.
../../rtl/fpga_core.v:175: warning: Port 20 (enable) of pcie_us_axis_cq_demux expects 1 bits, got 32.
../../rtl/fpga_core.v:175:        : Pruning (signed) 31 high bits of the expression.
../../rtl/fpga_core.v:175: warning: Port 21 (drop) of pcie_us_axis_cq_demux expects 1 bits, got 32.
../../rtl/fpga_core.v:175:        : Pruning (signed) 31 high bits of the expression.
../../rtl/fpga_core.v:229: warning: Port 8 (s_axis_tid) of axis_arb_mux expects 16 bits, got 32.
../../rtl/fpga_core.v:229:        : Pruning (signed) 16 high bits of the expression.
../../rtl/fpga_core.v:229: warning: Port 9 (s_axis_tdest) of axis_arb_mux expects 16 bits, got 32.
../../rtl/fpga_core.v:229:        : Pruning (signed) 16 high bits of the expression.
../../rtl/fpga_core.v:778: warning: Port 4 (s_axi_awaddr) of axi_ram expects 16 bits, got 32.
../../rtl/fpga_core.v:778:        : Pruning 16 high bits of the expression.
../../rtl/fpga_core.v:778: warning: Port 23 (s_axi_araddr) of axi_ram expects 16 bits, got 32.
../../rtl/fpga_core.v:778:        : Pruning 16 high bits of the expression.
../../rtl/fpga_core.v:835: warning: Port 8 (s_axis_tid) of axis_register expects 8 bits, got 32.
../../rtl/fpga_core.v:835:        : Pruning (signed) 24 high bits of the expression.
../../rtl/fpga_core.v:835: warning: Port 9 (s_axis_tdest) of axis_register expects 8 bits, got 32.
../../rtl/fpga_core.v:835:        : Pruning (signed) 24 high bits of the expression.
../../rtl/fpga_core.v:998: warning: Port 4 (s_axi_awaddr) of axi_ram expects 16 bits, got 32.
../../rtl/fpga_core.v:998:        : Pruning 16 high bits of the expression.
../../rtl/fpga_core.v:998: warning: Port 23 (s_axi_araddr) of axi_ram expects 16 bits, got 32.
../../rtl/fpga_core.v:998:        : Pruning 16 high bits of the expression.
../../rtl/fpga_core.v:1067: warning: Port 5 (cfg_interrupt_msi_vf_enable) of pcie_us_msi expects 8 bits, got 32.
../../rtl/fpga_core.v:1067:        : Pruning (signed) 24 high bits of the expression.
MODULE=test_fpga_core TESTCASE= TOPLEVEL=fpga_core TOPLEVEL_LANG=verilog \
        /usr/bin/vvp -M /usr/local/lib/python3.8/dist-packages/cocotb/libs -m libcocotbvpi_icarus   sim_build/sim.vvp -fst
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:74   in set_program_name_in_venv        Did not detect Python virtual environment. Using system-wide Python interpreter
     -.--ns INFO     cocotb.gpi                         ../gpi/GpiCommon.cpp:105  in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                         ..mbed/gpi_embed.cpp:244  in embed_sim_init                  Python interpreter initialized and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:202  in _initialise_testbench           Running on Icarus Verilog version 10.3 (stable)
     0.00ns INFO     cocotb                                      __init__.py:208  in _initialise_testbench           Running tests with cocotb v1.4.0 from /usr/local/lib/python3.8/dist-packages/cocotb
     0.00ns INFO     cocotb                                      __init__.py:229  in _initialise_testbench           Seeding Python random module with 1615876260
     0.00ns CRITICAL cocotb.regression                         regression.py:177  in _discover_tests                 Failed to import module test_fpga_core: No module named 'cocotb_test'
     0.00ns INFO     cocotb.regression                         regression.py:178  in _discover_tests                 MODULE variable was "test_fpga_core"
     0.00ns INFO     cocotb.regression                         regression.py:179  in _discover_tests                 Traceback: 
     0.00ns INFO     cocotb.regression                         regression.py:180  in _discover_tests                 Traceback (most recent call last):
                                                                                                                       File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 175, in _discover_tests
                                                                                                                         module = _my_import(module_name)
                                                                                                                       File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 69, in _my_import
                                                                                                                         mod = __import__(name)
                                                                                                                       File "/media/joy/D/cocotb/cocotb_work/cocotbext-pcie/verilog-pcie/example/ZCU106/fpga_axi/tb/fpga_core/test_fpga_core.py", line 28, in <module>
                                                                                                                         import cocotb_test.simulator
                                                                                                                     ModuleNotFoundError: No module named 'cocotb_test'
                                                                                                                     
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/cocotb/__init__.py", line 246, in _initialise_testbench
    regression_manager = RegressionManager.from_discovery(dut)
  File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 154, in from_discovery
    return cls(dut, tests, hooks)
  File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 126, in __init__
    for test in tests:
  File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 175, in _discover_tests
    module = _my_import(module_name)
  File "/usr/local/lib/python3.8/dist-packages/cocotb/regression.py", line 69, in _my_import
    mod = __import__(name)
  File "/media/joy/D/cocotb/cocotb_work/cocotbext-pcie/verilog-pcie/example/ZCU106/fpga_axi/tb/fpga_core/test_fpga_core.py", line 28, in <module>
    import cocotb_test.simulator
ModuleNotFoundError: No module named 'cocotb_test'
     0.00ns ERROR    cocotb.gpi                                gpi_embed.cpp:314  in embed_sim_init                  cocotb initialization failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:269  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
ERROR: results.xml was not written by the simulation!
make[1]: *** [/usr/local/lib/python3.8/dist-packages/cocotb/share/makefiles/simulators/Makefile.icarus:69: results.xml] Error 1
make[1]: Leaving directory '/media/joy/D/cocotb/cocotb_work/cocotbext-pcie/verilog-pcie/example/ZCU106/fpga_axi/tb/fpga_core'
make: *** [/usr/local/lib/python3.8/dist-packages/cocotb/share/makefiles/Makefile.inc:40: sim] Error 2

Please update me asap !!

joychauhan-sudo, 2021-03-16:
Thanks alex for quick response..
installed cocotb-test and cocotb-bus 
its working now...

alexforencich, 2021-03-18:
The missing cocotb-bus problem should also be fixed now as well; it didn't get added to the cocotb dependency list when cocotb 1.5.0 was released, so I added it as a dependency to cocotbext-axi.

closed 2021-05-21 by alexforencich