alexforencich/verilog-pcie #18: 64bit write/read instead of 32bit

joychauhan-sudo, 2021-08-24:
Hello Alex,

I want to write 64bit at a one time instead of 32bit write. 
Updated design as well as driver for 64bit using bar2. 
Can you please tell me what change i need to check for 64bit write/read. 
For now i'm using writel instead of iowrite32.

alexforencich, 2021-08-24:
Are you hoping to support using a 64-bit AXI lite interface, or to support 64-bit operations on a 32-bit AXI lite interface?  And are you using the UltraScale specific code (which is sort of deprecated at this point) or the newer modules with the generic PCIe interface?

alexforencich, 2021-10-03:
The `pcie_axil_master` module has now been updated to support arbitrary operations.  It still only supports a 32-bit AXI lite interface, but you can perform 64 bit reads and write and even unaligned reads and writes and it will handle them appropriately.

joychauhan-sudo, 2021-12-02:
Thanks for update

closed 2021-12-02 by joychauhan-sudo