alexforencich/verilog-pcie #21: Communication between logic and host using PCIe

noborienten, 2021-12-17:
Hi,

I want to deliver data generated by a specific logic to Host using PCIe. 
Can you advise on the necessary process for this?

alexforencich, 2021-12-17:
Well, it's not a strange question so much as an extremely general question.  There are several different ways you could go about doing this, depending on exactly what it is you want to do.  If you can provide some more specifics (what kind of data, where is it coming from, where is it going, how much BW, etc.), then I might be able to help you out.

If you're doing something networking-related, I suggest taking a close look at how Corundum is implemented, as this is the most comprehensive design available that uses the DMA components from this repo (well, this repo was built specifically to support Corundum in the first place).  But, this may not be exactly what you need for your application.

noborienten, 2021-12-17:
During 100 DRAM reads, I plan to implement logic to measure each DRAM read latency. 
I would like to send 100 latency data to the host CPU so that the driver of the host can output the data.

noborienten, 2021-12-17:
I think I asked an ambiguous question to answer because I lack basic knowledge of FPGA and PCIe. I am a graduate student in the master's course in computer science in Korea. I started studying fpga because I wanted to use fpga for research. I have completed studying verilog HDL. However, I don't know what to study about fpga design. There is no one to ask around, and there are not enough classes to learn. Can you teach me briefly what kind of study I have to do to understand your verilog-pcie code and design your fpga?

I'm sorry to ask you this question.

alexforencich, 2021-12-18:
No worries.  Also, if you want to discuss stuff feel free to hop on the Corundum slack workspace.  

Sounds like you want to implement some sort of a performance benchmark, correct?  These 100 DRAM reads would be out of host memory (via PCIe DMA), or out of on-card DRAM (via the Xilinx MIG, presumably over AXI)?  

Take a look at https://github.com/alexforencich/dma-bench; this repo has some logic to collect statistics for request latency for PCIe DMA.  The same logic is also implemented in Corundum.  The statistics counters can be read out via AXI lite, which is exposed in a PCIe BAR.  On my long-term to-do list is extending the DMA logic to support access to DRAM/HBM over AXI in addition to PCIe (well, the AXI DMA IF module is already in the repo, but the multiplexing logic to bind them together is not yet implemented).

noborienten, 2021-12-18:
I am planning to measure the performance of the on-card DRAM. As you said, I'll have to look at the code you told me.
Thank you so much for your guidance.

closed 2021-12-18 by noborienten

alexforencich, 2021-12-18:
In general, if your data transfer requirements are small and you don't need to worry about performance, latency, etc., then just exposing the data via registers over AXI-lite is going to be by far the simplest thing to do.  No need to make it any more complex than it needs to be.  

Also, I unfortunately don't have any good pointers for where to learn this stuff...I'm mainly self-taught, just by diving in and figuring things out.  For this repo, I spent a long time looking at the PCIe IP core manuals, the PCIe spec, testing things out on the FPGAs, and looking at things with the ILA.  For Corundum, I spent quite a long time reading over the Mellanox device driver source code and the Exablaze open-source repo.

noborienten, 2021-12-18:
Actually, my purpose is not to measure performance. I am in the process of testing whether a cover channel using DRAM bandwidth saturation is possible at FPGA.

Your answer is a great help to me. I respect your efforts in this repo and Corundum. Thank you for explaining the process of self-taught, and I will try to do it like you.