alexforencich/verilog-pcie #27: Send immediate data to host?

filamoon, 2022-01-22:
I wonder what is the best way to send a small amount of immediate data (say 4 bytes data not in the ram) to the host?
Would it be good to add s_axis_write_desc_imm_data/s_axis_write_desc_imm_data_enable to dma_if_pcie_wr so it can bypass ram read?

alexforencich, 2022-01-22:
I have had the same thought myself, and am considering adding a full AXI lite interface to the DMA engine for doing small reads and writes against host memory. Doing writes would be particularly useful for corundum so queue pointers can be maintained directly in host memory to avoid MMIO reads. Haven't had time to sort out all of the details yet, though. But I suppose I could probably start with just the write side since that would be significantly simpler.

filamoon, 2022-01-22:
If you can provide some hints maybe I can do some help :)

alexforencich, 2022-01-22:
Unfortunately, it doesn't look like it's going to be a super straightforward change.  The current write DMA engine has a flow to it that goes something like this:

1. request is accepted, op table entry is allocated, request is segmented into TLPs
2. internal reads are issued for each TLP
3. TLP header is generated and payload is assembled from read data
4. TLP transmission is confirmed via TX sequence number, operation is reported as complete when the last TLP is sent

So, this results in a bit of a conundrum.  How should the immediate data be handled?

One option is to accept it in step 1 and allocate an op table entry for it, same as how a new request is handled.  The immediate data would be stored in the op table.  But no internal memory reads will be issued, so there will be no read data to use for the payload in step 3.  But possibly this is not a big deal.  This might actually be the best way to go, as I suspect some relatively complex logic will be needed to deal with wstrb properly, particularly for a 64-bit wide AXI lite interface.

Another option is to accept it in step 3 and directly generate a TLP without adding an entry in the op table.  One advantage here is potentially lower latency, and also skipping the "queue" of DMA write requests.  And also potentially also lower resource consumption, as the data does not need to be stored in the op table.  However, the resulting TLP will have to be tracked somehow so the completion can be correctly reported over AXI lite.  Not sure the best way to do that.

In the read direction, the situation is somewhat similar, but there will probably need to be additional logic (basically a FIFO of sorts) to store each in-progress AXI lite operation so the results can be returned in-order.  Possibly this would be an additional op table, but for AXI operations only.  And I think a bit would need to be added to the tag table to keep track of AXI lite operation vs. normal DMA read.

Although now that I think about it some more, maybe implementing AXI lite itself is not such a great idea, since AXI lite requires aligned operations and this could complicate writing small bits of data to unaliged locations in host memory.  Possibly it makes more sense to implement something more like what you're suggesting, tacking a signal for immediate data to the request descriptor.  And that could potentially also be used in the read direction, returning the data with the DMA status response.  

Another consideration is that MSI/MSI-X on certain devices is also implemented as explicit PCIe write requests, instead of a separate signal to the hard IP core that results in a write request being generated.  Another potential option is to merge in these small writes deeper in the pipeline, possibly in the interface shim instead (`pcie_us_if_rq`/`pcie_s10_if_tx`).

alexforencich, 2022-04-04:
Added this capability in https://github.com/alexforencich/verilog-pcie/commit/a5dcb3d27c0330c8bdecf372aa431d1fcc83bb1b

closed 2022-06-03 by alexforencich