alexforencich/verilog-pcie #28: unexpected dma read request logged

filamoon, 2022-02-16:
From the log I see one request:

#   5618.12ns INFO     Memory read, address 0x00001000, length 28, BE 0xf/0xf, tag 30

However, I logged each request on dma_if_pcie_rd like this and found no such request at all.

    always @(posedge clk) begin
        if (s_axis_read_desc_valid && s_axis_read_desc_ready)
            $display("[%0t] dma_if_pcie_rd: read dma_addr %0h len %0h", $stime,
                s_axis_read_desc_pcie_addr,
                s_axis_read_desc_len
            );
    end

No address 0x00001000 or length 28 logged.
What could go wrong?

alexforencich, 2022-02-16:
Well, there isn't a 1:1 correspondence between transfer requests issued to the DMA engine, and the actual PCIe read requests that the DMA engine generates.  Requests will be split under several conditions, but you're not providing enough information for me to be able to say exactly what happened.  However, I suspect that you may have issued a transfer request that crosses a 4K address boundary and this is the second half of that transfer request, as 0x1000 is a 4K boundary and PCIe read and write request cannot cross 4K boundaries.

alexforencich, 2022-02-16:
Also, I should mention that the length reported there is the length field from the TLP, which is reported in DWORDs.  So in this case, 28 DW with first/last BE of 0xF/0xF is 112 bytes.

filamoon, 2022-02-16:
thanks a lot for the quick reply! I found the root cause request with your help.

closed 2022-03-31 by alexforencich