alexforencich/verilog-pcie #33: Why there is no sgDMA in example_core.v

chenbo-again, 2023-06-18:
Hi Alex, I get a lot from video and your project Corundum. 
I read Corundum and this repo for days and I found that many code in corundum is about sgDMA. And there is only a simple block DMA  in verilog-pcie repo.
In fact, I plan to impl a sgDMA for a open source rdma, I wonder why you do not integrate sgDMA in example_core, but place it in corundum. That is very different from riffa.
Is there any cons to do so?

alexforencich, 2023-06-18:
The example designs here are really not much more than sanity checks and some simple benchmarks. There is no descriptor handling, etc. in this repo. Scatter/gather is really more a feature of the descriptors than the DMA engine itself. With the split DMA interface / DMA client architecture, scatter/gather simply means issuing multiple operations to the DMA interface for one operation on the DMA client.

For open-source RDMA, what about helping to implement that within Corundum, instead of building a separate project that's necessarily going to reimplement a lot of stuff that already exists in Corundum?

alexforencich, 2023-06-18:
Also, I should mention that the descriptor format and management logic is necessarily going to be application-dependent, hence why Corundum has its own implementation. XDMA, QDMA, RIFFA, etc. all have serious limitations due to their descriptor formats. Sure, you can do host-managed memcpy effectively with all of those, but what about per-packet PTP timestamps, checksum offloading, etc.? All of that requires customization of the descriptors and completions. So while I could put some sort of descriptor management logic in this repo, Corundum would not use it, and you probably wouldn't want to use it as-is either. (and if corundum doesn't use it, I don't have time to deal with it....). RDMA in particular really needs a custom descriptor format to encode all of the metadata and parameters associated with each operation.

chenbo-again, 2023-06-18:
> Also, I should mention that the descriptor format and management logic is necessarily going to be application-dependent...... RDMA in particular really needs a custom descriptor format to encode all of the metadata and parameters associated with each operation.

Thank you, that's what I need.

For open source RDMA, I know little about this project, and what I need to do is provide a reference design to replace XDMA, I think I will refer to corundum (both idea and code). but implement that within Corundum, this needs further discussion. Thank you for your help.

closed 2023-06-18 by chenbo-again