alexforencich/verilog-pcie #39: Implementation of PCIe “Non-Transparent Bridge" (NTB)

ohault, 2023-08-29:
PCIe interconnect with multiple PCIe Endpoints and hosts (FPGA, DPU, CPU, GPU, ...) can be organized in several topologies. 

A NTB implementation for FPGA will provide the cross-domain communication support at PCIe level without relying on ETH PHY and the way to support these others topologies.

Basseuph, 2023-08-31:
Hello Olivier,

it would be really interesting to have such a thing publicly available, although I guess it would make sense to wait for the PCI-SIG to finalize the effort including this topic.

However, in the meantime you might be interested in an NTB [1] available for a couple of FPGA platforms, see also [2].

Ulrich

[1] https://www.missinglinkelectronics.com/pci-express-pcie-connectivity/
[2] https://www.missinglinkelectronics.com/wp-content/uploads/2019/06/04_04_Sensor-Fusion-and-Data-in-Motion-Processing-for-Autonomous-Vehicle.pdf

ohault, 2023-08-31:
> Hello Olivier,
>
> it would be really interesting to have such a thing publicly available, although I guess it would make sense to wait for the PCI-SIG to finalize the effort including this topic. [...]

Hi @Basseuph, thank you for this presentation.

My understanding is that there is a lack of accurate and shared understanding of PCIe specs and capabilities.
I just recently asked PCI-SIG to define a free Adopter type membership like in the CXL Consortium.

Today, I guess we can consider that FPGA development is no longer a niche market for hardware engineers and that NTB is the cornerstone to unlock the potential of an open source ecosystem.

MLE may have a key role to play here.