alexforencich/verilog-pcie #50: why I run the  test instance of tb , it report cannot import module dma_ram?

zhfff4869, 2024-05-02:
[image]

alexforencich, 2024-05-02:
Likely a broken symbolic link.

zhfff4869, 2024-05-02:
> Likely a broken symbolic link.
I use `tox -e py39` , then it can run, however many cases is failed. I wonder where is the problem?

[image]
[image]
[image]
[image]
[image]
[image]