alexforencich/verilog-pcie #52: Driver and Verilog mismatch

alexisfrjp, 2024-05-13:
Driver not compatible with the verilog code?

Driver:

```
static void dma_cpl_buf_test(struct example_dev *edev, dma_addr_t dma_addr,
		u64 size, u64 stride, u64 count, int stall)
{
	unsigned long t;
	u64 cycles;
	u32 rd_req;
	u32 rd_cpl;

	rd_req = ioread32(edev->bar[0] + 0x000020);
	rd_cpl = ioread32(edev->bar[0] + 0x000024);

	// DMA base address
	iowrite32(dma_addr & 0xffffffff, edev->bar[0] + 0x001080);
	iowrite32((dma_addr >> 32) & 0xffffffff, edev->bar[0] + 0x001084);
	// DMA offset address
	iowrite32(0, edev->bar[0] + 0x001088);
	iowrite32(0, edev->bar[0] + 0x00108c);
	// DMA offset mask
	iowrite32(0x3fff, edev->bar[0] + 0x001090);
	iowrite32(0, edev->bar[0] + 0x001094);
```

I see `0x20`... and lots of `0x1xxx` addresses but they don't correspond to anything in the verilog code:

[image]

Why?

alexforencich, 2024-05-13:
https://github.com/alexforencich/verilog-pcie/blob/master/example/common/rtl/example_core.v#L522-L523 etc.

alexforencich, 2024-05-13:
If you're asking about the AXI example designs, those are more or less deprecated.

alexisfrjp, 2024-05-13:
Yes, `fpga_axi` examples (https://github.com/alexforencich/verilog-pcie/blob/master/example/VCU118/fpga_axi/rtl/fpga_core.v#L485)

Ok, thank you, good to know, you should maybe remove them from master?

closed 2024-05-29 by alexisfrjp

alexisfrjp, 2024-07-05:
Ok now I understand why AXI modules are deprecated, they are completely wrong and mess up the us+ core by not keeping tvalid high for a packet. It works only if the AXIMM is connected to a bram.

You should completely remove them from the project.