alexforencich/verilog-pcie #55: Whether PCIe Gen4/Gen5 is supported？

Jzone315, 2024-07-03:
Hi，thanks for this great achievement，i see there  is an reference design for intel DE10_Agilex，and i had restored it with Quartus，but its PCIe rate is still Gen3x16，so i wonder if this design supports PCIe Gen4/Gen5 or not？

alexforencich, 2024-07-03:
It does not currently close timing at the clock frequencies required. With the P tile, gen 4 x16 runs at 450-500 MHz. On the Agilex I series with the R tile, it might be possible to support gen 4 x16 as that runs at 1024 bits at 250 MHz. But the R tile is currently not supported as the flow control is different from the P tile. I'm planning on looking in to this relatively soon, but unfortunately it won't help with the P tile at all.

Jzone315, 2024-07-05:
Ok, I see，thanks a lot for the reply.

closed 2024-07-05 by Jzone315