alexforencich/verilog-pcie #58: Link status has only Gen1*16

libero-x, 2024-10-07:
hello，Alex,when I used the VCU118,it's Link status just has only Gen1*16,I don not know how to solve it.

[image]

alexforencich, 2024-10-07:
what are the capabilities of the upstream port?  (see `lspci -vvv -s 0000:00:01.0`)

libero-x, 2024-10-08:
I think it is Gen3*16,this is the output of command "ls pci" 
[image]

alexforencich, 2024-10-08:
Have to run lspci -vvv as root to see all of the capability info, specifically the link cap and link status registers

libero-x, 2024-10-09:
Thank you for reply!This is the detail of "sudo lspci -vvv".Maybe the reason is ASPM disabled? 
[image]

alexforencich, 2024-10-09:
Probably not power management, dynamically changing the link speed I think is sort of a custom power saving technique that has to be implemented on the device.  I suspect either signal integrity or equalization.  Are you plugging the card straight into the slot, or are you using an extender?  First thing to do is try it without an extender, second thing to do is to force BIOS to fully reinitialize the link by removing the card, booting the system with the slot completely empty, shutting it down, reinstalling the card, and trying again.

libero-x, 2024-10-17:
hello,Alex,I plugg the card straight into the slot before.I have attemped the methed that you give me,but not useful.So I am trying the XDMA ip to accomplish the PCIe,but it's "lspci -vvv"give me 5 lane Err as the following. And I can write or read data from the VCU118 card.The speed only have 8GBps,but the Gen3*16 should >14Gbps. Could you give me some help?
[image]

alexforencich, 2024-10-17:
Sorry, I'm not familiar with XDMA.  But it uses the same transceivers and hard core, so if that enumerates at gen 3 x16, then the example design should also work at that speed.  I'm not sure why you're seeing lane errors, possibly this indicates some sort of signal integrity issue.