alexforencich/verilog-pcie #62: test_pcie.py:26: in <module>     from myhdl import * E   ModuleNotFoundError: No module named 'myhdl'

kjhhgt76, 2025-03-04:
OS: Windows 11 WSL Ubuntu 24.04.1 LTS

Simulator: Icarus Verilog version 13.0 (devel) (s20250103-25-g99580cd05)

Python library:
cocotb         1.9.2
cocotb-bus     0.2.1
cocotb-test    0.2.6
cocotbext-axi  0.1.24
cocotbext-pcie 0.2.14
find_libpython 0.4.0
iniconfig      2.0.0
packaging      24.2
pip            24.0
pluggy         1.5.0
pytest         8.3.5
setuptools     75.8.2

Step to reproduce:
1. cd to verilog-pcie/tb
2. Run the following command:
SIM=icarus pytest -o log_cli=True test_pcie.py

Output:
\================================================================================================================================= test session starts ==================================================================================================================================
platform linux -- Python 3.12.3, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/yorkt/verilog-pcie
configfile: tox.ini
plugins: cocotb-test-0.2.6
collected 0 items / 1 error

\======================================================================================================================================== ERRORS ========================================================================================================================================
___________________________________________________________________________________________________________________________ ERROR collecting tb/test_pcie.py ___________________________________________________________________________________________________________________________
ImportError while importing test module '/home/yorkt/verilog-pcie/tb/test_pcie.py'.
Hint: make sure your test modules/packages have valid Python names.
Traceback:
test_pcie.py:26: in <module>
    from myhdl import *
E   ModuleNotFoundError: No module named 'myhdl'
\=============================================================================================================================== short test summary info ================================================================================================================================
ERROR test_pcie.py
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
\=================================================================================================================================== 1 error in 0.09s ===================================================================================================================================

possible cause:
myhdl is required 

possible solution:
python library "myhdl" should be added to the requirement.

kjhhgt76, 2025-03-04:
It seems that myhdl is not installed along with cocotb. Thus it needs to be installed separately.

alexforencich, 2025-03-04:
Those are the old MyHDL testbenches.  The current cocotb ones are in the subdirectories.

closed 2025-03-04 by alexforencich

kjhhgt76, 2025-03-04:
I see. Then I will use the subdirectories from now on.
Please remove the old tb if they are the subset of the new tb.

alexforencich, 2025-03-04:
This repo is being deprecated, to be replaced by https://github.com/fpganinja/taxi which will have only the new stuff.

kjhhgt76, 2025-03-04:
I see. But the new one is still under development. So I will try this one first.